diqezit/ats20_ats_ex #1: CW LSB/USB Tone Mismatch and Minor UI Concerns

jh4vaj, 2025-07-02:
I noticed a slight frequency offset between CW LSB and CW USB modes. Here’s how I tested it:

1. Set the ATS-20+ as follows:
    - MODE: CW
    - CW: LSB
    - BFO: 0
    - FREQ: 7100 kHz
1. Output a 7100 kHz signal from a signal generator.
1. Adjust the SG frequency until the tone heard from the ATS-20+ is approximately 500 Hz.

In this configuration, the closest tone to 500 Hz was actually around 495 Hz.

Next, I switched to CW USB and tuned the receiver to 7100 kHz again. 
This time, the audible tone was about 398 Hz. I was expecting to hear 500 Hz, so it was off by roughly 100 Hz.

This level of offset might be considered acceptable, but I wanted to report it just in case.

Also, when switching between CW LSB and CW USB, the tuning step is automatically forced to 10 Hz, 
and the frequency shifts—possibly to the nearest grid point? 
It's a bit inconvenient that just changing the sideband mode alters both step size and frequency.

Two more unrelated items:

Has the bar-style S-meter display been removed? On FM, the RSSI is shown as a numeric value, 
but there’s no signal indication at all on MW/SW. It feels slightly inconvenient.

Lastly, regarding the FM station memory function: 
I suggest moving the description of “Save Current Station” to the top of the menu
—it would make the process easier to understand. 
Alternatively, separating the instructions for saving a station and for recalling it afterward might help. 
At first, I mistakenly thought I needed to open the Favorites Menu in order to save a station.

[image]

jh4vaj, 2025-07-02:
I set the BFO to 0 to keep things simple. 
Then I adjusted the signal generator’s frequency so that the audio tone would be 500 Hz under that condition.

The main point I wanted to raise is that, even when the reception frequency is the same, 
the audible tone changes between CW LSB and CW USB modes. 
If we start adjusting the BFO as well, things can get a bit more complicated—at least that’s how I see it.

Apologies if I’ve misunderstood anything.

diqezit, 2025-07-02:
Thank you for the detailed test results. 

For now your findings, combined with the analysis from @G8PTN ([here](https://github.com/goshante/ats20_ats_ex/issues/42#issuecomment-3015265184)), have made the root cause of the problem clear

The main issue is that the current **100Hz BFO calibration step is too coarse** to compensate for the observed 75Hz error...

A change here that implements a finer 10Hz calibration step. 

Could you please test how the receiver works after compiling with the code below?

```cpp

void updateBFO() {
    int16_t finalBfo = g_currentBFO + (g_Settings[BFO].param * 10);

    if (g_currentMode == CW) {
        if (g_Settings[CWSwitch].param == 1) { // 1 = USB
            finalBfo -= CW_PITCH_OFFSET_HZ;
        }
        else { // 0 = LSB
            finalBfo += CW_PITCH_OFFSET_HZ;
        }
    }

    g_si4735.setSSBBfo(finalBfo * -1);
}

void doBFOCalibration(int8_t v) {
    doSwitchLogic(g_Settings[BFO].param, -125, 125, v);

    if (isSSB()) {
        updateBFO();
    }
}
```

After applying these changes, try changing the BFO settings and set the values (think `-8` )

It would be great if you could provide new tests results

diqezit, 2025-07-02:
> Has the bar-style S-meter display been removed?

Yeah, was removed during to save flash space

> On FM, the RSSI is shown as a numeric value, but there’s no signal indication at all on MW/SW. It feels slightly inconvenient.

Agree! My original plan was to add a proper S-meter for all modes, not just the numeric RSSI for FM..

Unfortunately, ran into the flash memory limit of the ATmega328P. To ensure all working, had to postpone this feature

My plan for the next  version (v3.4) with focus on code optimization (eeprom OFC). If I can free up enough flash memory, I will definitely implement a proper S-meter for **AM, SW, and SSB modes**. I'll also add an option in the settings menu to toggle it on/off, just like the numeric RSSI for FM

a graphical S-meter for SSB would be useful too as well, right? (Well, that's after just the next fix you specified above)

jh4vaj, 2025-07-03:
I applied the source code changes as suggested. 
Just like before, I set the signal generator to output 7100 kHz and tuned the ATS-20+ to 7100 kHz.

With BFO set to 0, I observed the following tones:

- CW LSB: 575 Hz
- CW USB: 325 Hz

These results are identical to those from before the code modifications. 
I'm not sure if it makes sense to measure other frequencies at this point, so I'm reporting this first.

The issue where switching between CW LSB and USB unintentionally altered the frequency step has been resolved.

By the way, the signal generator I'm using is LiteVNA. It outputs a single fixed frequency in CW FREQ mode. 
With LiteVNA set to 7100 kHz, a separate frequency counter measured 7099.992 kHz, so I believe the accuracy is reasonably good.

NanoVNA should support similar functionality, and tinySA also has a signal generator feature. Hope that's useful for reference.

jh4vaj, 2025-07-03:
One more note to add: 
Even after applying the previous changes, the BFO step size remains 100 Hz. 
Each increment of BFO by 1 results in a 100 Hz shift in the audible tone. 
I had forgotten to mention this important detail, so I’m adding it here as a supplement.