diqezit/ats20_ats_ex #2: Remember last tuned frequency in each band

radiola2, 2025-07-03:
Could it be made that it remembers the last frequency tuned in each band?

diqezit, 2025-07-03:
Hello,

I'd like to clarify the details on the feature request about remembering the last frequency

Am I correct in understanding that you would like the firmware to save the state for each individual SW sub-band?

Currently, in firmware version 3.3, the system remembers the last active frequency for the entire SW range as a whole, but not for each specific element within the `SWSubBands` array (like 40m, 20m, etc.)

If my understanding is correct, the implementation would require the following changes:

1.  **The `bandSwitch` function would need to be reworked.** Instead of just jumping to the next predefined frequency from `SWSubBands`, its new logic would be:
    *   **Before leaving** the current sub-band (e.g., 40m), it must **save** the user's currently tuned frequency (e.g., 7125 kHz) to a memory slot associated with the 40m band
    *   **When switching** to a new sub-band (e.g., 31m), it must **load** the previously saved frequency from the slot for 31m (or a default frequency if it hasn't been visited before)

2.  **All of these saved frequencies for each sub-band would need to be stored in EEPROM** to ensure they persist after the device is powered off

to summarize, am I right that the goal is to transform the current "quick bookmark navigation" into a full "state-saving system for each individual sub-band"?

If this is what you envisioned, I will proceed with this implementation.

Thank you

diqezit, 2025-07-03:
You can check out the test release below

Per-Band State Saving
Unified Band Switching
Seamless Band Coverage

[ATS_EX.zip](https://github.com/user-attachments/files/21047693/ATS_EX.zip)

diqezit, 2025-07-04:
The problem is caused by recent architectural changes, specifically splitting the sw entity into sub-bands for data persistence , so this important change affected a significant part of the code base, especially with processing, leading to unforeseen errors

The current firmware version is unstable (test) as I wrote earlier. Later plan to return to this task to eliminate the bug and finalize this to v4.0

radiola2, 2025-07-11:
Yes! Now it's much better.

But there is something strange happening when I rotate the knob very fast on AM. The display shows that the frequency is changing, but the received frequency can stay on the same spot for several seconds if you spin it at the "right" speed. Is this because of the quite low end CPU or is something else going on? I tried the original Goshante firmware, and the behavior partly showed up there too, but not as much as in this mod version (not sure but I think so).

With this behaviour you can easily "miss" stations when you're tuning.
Otherwise it's getting very much better.

closed 2025-07-14 by diqezit