diqezit/ats20_ats_ex #3: SSB Demodulation Not Working in v4.2

jh4vaj, 2025-07-06:
As stated in the title, I tried v4.2 and found that SSB demodulation is not functioning.

diqezit, 2025-07-06:
did you have turn the RSI (RSSI) indicator on in sync/ssb mode?  
Potentially, the only problem may lie in the updateSignalQuality function and nowhere else. There to update the RSSI on lcd to there need push the update through an internal command which gives interference in AM reception - you can disable it in RSI settings - off for now

Yes, it is possible to make the greeting more informative in accordance with the modification. i do

jh4vaj, 2025-07-06:
It seems the RSSI indicator is now available in AM/SSB modes—I hadn’t noticed that. 
I found the setting in the menu; it was set to “AUT” (probably short for “AUTO”). 
I switched it to “On,” but it didn’t resolve the SSB demodulation issue. 

Currently, turning the VFO dial in SSB mode has almost no effect on the audio. 
So the problem may not be just that SSB demodulation fails—it looks like SSB mode itself might not be functioning properly.

diqezit, 2025-07-06:
Thank you, too, for the detailed feedback!

I have carefully studied all your comments and want to provide a detailed response for each point, including some technical details about the decisions I made, to avoid any misunderstandings.

**1. On RSSI Display and Audio Interference ("Clicking")**
> Just to confirm-RSSI isn't displayed in SSB mode, right?

In SSB mode, the RSSI is no longer displayed (as of v4.3), and for AM mode, there is now an option to disable it. This was done intentionally to improve audio quality, as the problem turned out to be deeper.

Discovered that the process of requesting the signal level from the Si473x chip over the I2C bus in AM/SSB mode creates a small but very unpleasant load. This manifested as a periodic "clicking" sound that followed the IC data update timer, which was especially disruptive when listening to weak stations overall

* Therefore, added an option in the settings menu (`RSSI AM Off`) that allows the user to decide what is more important: seeing the signal indicator or having the cleanest possible audio.
* In FM mode, this problem is absent, so the RSSI indicator works stably and constantly.
* To minimize the impact on the audio even when RSSI is enabled, I also implemented a special timer for AM. After any user action (e.g., turning the encoder), a data update request is sent once, and then its display is "frozen" until your next interaction. This prevents constant requests to the chip from creating interference while you are listening.

`
// logic for updating the signal quality indicator
static inline void updateSignalQuality() {
    uint8_t new_value;

    // Disable RSSI query in SSB mode to prevent interference with the SSB patch
    if (isSSB() || g_currentMode == CW) {
        new_value = 255;
    }
    else if (g_currentMode == FM) {
        g_si4735.getCurrentReceivedSignalQuality(1);
        new_value = g_si4735.getCurrentRSSI();
    }
    else { // AM mode
        if (g_Settings[RSSI_AM_Off].param == 1) {
            new_value = 255;
        }
        else {
            if (g_forceRssiUpdate) {
                g_si4735.setFrequency(g_currentFrequency);
                g_forceRssiUpdate = false;
            }
            g_si4735.getStatus();
            new_value = g_si4735.getReceivedSignalStrengthIndicator();
        }
    }

    // Cap value at 99 for the 2-digit to overlow on scren (only 2 chars not 3)
    new_value = (new_value > 99 && new_value != 255) ? 99 : new_value;

    if (g_signalQualityValue != new_value) {
        g_signalQualityValue = new_value;
        showSignalQuality();
    }
}
`

**2. The VFO "Jump" on Default Frequency**
> when tuning across SW sub-bands using the VFO, it jumps to the sub-band's default frequency... this behavior only occurs the first time you enter a sub-band

As you correctly noted, firmware now has a "seamless transition" mechanism between bands. Its main purpose is  automatically load the correct **step** and **bandwidth (bw)** settings that are saved for that new band. 

Its current has one side effect - as example so when you are tuning up from 7299 kHz to 7300 kHz, firmware sees that this is a new sub-band ("41m"). It looks into its memory (EEPROM) and sees the *default* frequency stored for the "41m" band (e.g., 8500 kHz from `g_bandList`). At this moment, my current logic loads this default frequency, instead of staying at 7300 kHz. This is precisely why the "jump" occurs.

After the initial jump, tuning to a new frequency (e.g., 8600 kHz) engages the "idle save" mechanism. Its function is to write the final frequency to EEPROM, but only after a 30-second timer of user inactivity expires (SAVE_ON_IDLE_TIMEOUT). I increased this delay from the original 10 seconds. 
With a 10-second delay, the system would often save an intermediate, not the final frequency. 
This was especially noticeable when working with weak signals - when  makes a micro-adjustment, then stops to listen, and an unwanted save would occur at that moment...
The 30-second interval gives the user enough time to evaluate a signal.

This single write operation overwrites the band's default frequency in memory with the new one (8600 kHz). This is why on all subsequent entries into that band, the jump is to the last-saved frequency, not the initial default.

This is exactly why, if you leave this band and re-enter it, the firmware will "jump" again, but this time not to the default frequency, but to your *last saved one*. This is less noticeable and feels more logical, which is why it seems that the problem has disappeared.,

**3. Saving Frequency when Switching Between LSB and USB**
> It also looks like the frequency isn't saved when switching between LSB and USB...
[... 20 lines cut ...]
    { PACK_STR4("SW  "),  5060,      5900, SW_BAND_TYPE,  5500,        1,      4,       1,      4,    4,     0 },
    { PACK_STR4("49m "),  5900,      6200, SW_BAND_TYPE,  6000,        1,      4,       1,      4,    4,     0 },
    { PACK_STR4("SW  "),  6200,      7000, SW_BAND_TYPE,  6500,        1,      4,       1,      4,    4,     0 },
    { PACK_STR4("40m "),  7000,      7300, SW_BAND_TYPE,  7150,        1,      4,       1,      4,    4,     0 },
    { PACK_STR4("41m "),  7300,      9400, SW_BAND_TYPE,  8500,        1,      4,       1,      4,    4,     0 },
    { PACK_STR4("31m "),  9400,      9900, SW_BAND_TYPE,  9600,        1,      4,       1,      4,    4,     0 },
[... cut ...]
Therefore, in the next update, I will not be able to implement separate frequency saving for LSB and USB. This remains a known limitation. To return to this issue in the future, I will need to find ways to optimize memory usage, for example, by reducing and optimizing the code. **It might be possible to use bitwise arithmetic** to pack more data into the same space, but this is a complex solution that requires time.

---
Thanks again for your invaluable help! I hope my response was clear and logical :)

jh4vaj, 2025-07-07:
I’ve confirmed SSB reception in v4.3—everything seems to be working fine.

Now, about switching between LSB and USB in CW mode: as mentioned in @G8PTN ’s earlier post, 
this is to avoid interfering signals. I’ll re-share that diagram for reference. 

[image]

The problem is, if switching LSB/USB changes the receive frequency, 
it can cause you to lose track of the target signal. Ideally, 
you’d always hear the same tone defined by CW_PITCH_OFFSET_HZ, regardless of LSB or USB. 
Of course, I understand it might be asking too much from this receiver. 
But since this behavior worked correctly in v3.4, the change is a bit disappointing.

As for EEPROM limitations—this is a bit of a workaround, 
but switching the board setting from Nano to UNO would increase available space, right? 
Even now, I compile using the UNO setting in the Arduino IDE, as it no longer fits under Nano. 
That said, this would require a Nano with an Optiboot bootloader, so I suppose it’s more of a last resort.

diqezit, 2025-07-07:
> The problem is, if switching LSB/USB changes the receive frequency,
> it can cause you to lose track of the target signal. Ideally,
> you’d always hear the same tone defined by CW_PITCH_OFFSET_HZ, regardless of LSB or USB.

Yes, thank you. And my statement about eeprom was not correct - the emphasis is now on flash memory - as the compiler will be able to interpret the logic more compactly.