diqezit/ats20_ats_ex #8: Frequency Rounding Behavior in Quick Sync Toggle Function

jh4vaj, 2025-07-10:
When receiving a signal such as 10000.10 kHz in SSB or CW mode, 
long-pressing the MODE button causes the frequency to be rounded to 10000.00 kHz.

Now, regarding the display behavior: 
The Quick Sync Toggle function is intended to toggle the receive sideband, correct? 
For example, the display switches from LSB to LSB S, but wouldn’t it make more sense for this to change to USB instead? 
Similarly, USB S could be replaced with LSB. 
In the case of CW, many users might find CW R (Reverse) more intuitive than CW S.

diqezit, 2025-07-10:
>The Quick Sync Toggle function is intended to toggle the receive sideband, correct?

**long** pressing the MODE button will enable/disable the ssb mode, while **short** pressing the MODE button will switch between LSB/USB. In the original firmware from the author it was possible to switch SYNC mode by long pressing the AGC button.

yes - rounding off g_currentBFO = 0;
So that you get a VFO reset to .00.

comment out g_currentBFO = 0; in configureSSBMode

> In the case of CW, many users might find CW R (Reverse) more intuitive than CW S

I don't know about that. I'm not much of an SSB fan.
But in CW SYNC works why not?
It would not be economical from the point of view of code optimization by the compiler, at least on 4.8 where only 2 bytes are free.

jh4vaj, 2025-07-11:
Sorry—I had misunderstood the functionality. 
This is not for selecting sidebands, but rather for enabling or disabling the synchronous detection feature, correct? 
If that’s the case, then rounding off the decimal portion of the frequency seems like appropriate behavior. 
Also, in that case, I don’t think synchronous detection is necessary in CW mode, as it’s primarily intended for AM or DSB.

- Here’s the behavior I’d ideally like to see:
- MODE short press: AM → SSB → CW → AM → SSB ...
- MODE long press: Toggle synchronous detection on/off
- STEP long press: Toggle LSB/USB (in SSB, sync detection, and CW modes)

I’m not sure if this can actually be implemented, but I feel the operation would be more streamlined this way.

diqezit, 2025-07-19:
check 5.1 - navigation according to your requirement, if it is OK - close the issue

jh4vaj, 2025-07-19:
Thank you for implementing those features. Overall, they work well, but there are a few areas that could be improved:

- Sync detection in CW mode (MODE long press) is unnecessary
- In CW mode, I’d like to be able to switch between LSB and USB using STEP long press, just like in SSB mode

I’d appreciate it if you could consider these points. Thank you.

diqezit, 2025-07-21:
Hi. Just wanted to let you know that I took your recommendation. And have made the corrections in the new version. If you have a chance let me know the results and we can close the Issue