diqezit/ats20_ats_ex #16: How to save 50 bytes on the battery indicator

den3rats, 2025-07-21:
Hi!
We can save 50 bytes flash for your great plans.
we can use piecewise linear interpolation instead of a table for the battery indicator.

Part1. Remove declaration:
```
static const BatteryPointCompressed battery_table[] PROGMEM = {
    // Offset, Percent, (ADC -> Approx. Voltage)
    {156, 100}, // 647 -> 4.15V
    . . .
    {  0,  0}   // 491 -> 3.15V (Safe cut-off)
};
```

Part2. Piecewise linear interpolation instead of table usage.
Dirty but works:
```
// Calculates raw battery percentage from an ADC value using interpolation
// It was a table with conversion in original;
// If analize it as lines, you can find two point with significant angle change.
// And with almost linear chunks between them.
// This allows you to use piecewise interpolation instead of a tabular description.
//    // Offset, Percent, (ADC -> Approx. Voltage)  tg(angle)
//    {156, 100}, // 647 -> 4.15V                   100 / 73
//    {148, 90},  // 639 -> 4.10V                   85 / 133
//    {141, 85},  // 632 -> 4.05V                   85 / 133
//    {132, 80},  // 623 -> 4.00V                   85 / 133
//    {125, 75},  // 616 -> 3.95V                   85 / 133
//    {117, 70},  // 608 -> 3.90V                   85 / 133
//    {109, 65},  // 600 -> 3.85V                   85 / 133
//    {101, 60},  // 592 -> 3.80V                   85 / 133
//    { 93, 55},  // 584 -> 3.75V                   85 / 133
//    { 86, 50},  // 577 -> 3.70V                   85 / 133
//    { 78, 45},  // 569 -> 3.65V                   85 / 133
//    { 70, 40},  // 561 -> 3.60V                   85 / 133
//    { 62, 35},  // 553 -> 3.55V                   85 / 133
//    { 55, 30},  // 546 -> 3.50V                   85 / 133
//    { 47, 25},  // 538 -> 3.45V                   85 / 133
//    { 39, 20},  // 530 -> 3.40V                   85 / 133
//    { 31, 15},  // 522 -> 3.35V                   85 / 133
//    { 23, 10},  // 514 -> 3.30V                   85 / 133
//    { 15,  5},  // 506 -> 3.25V                   85 / 133
//    {  7,  2},  // 498 -> 3.20V                   5 / 15
//    {  0,  0}   // 491 -> 3.15V (Safe cut-off)    5 / 15
static uint8_t calculateRawPercent(uint16_t adc_value) {
    if (adc_value >= 647) return 100;
    if (adc_value <= 491) return 0;

    if (adc_value < 506) {
        return      (adc_value - 491) * 5 / 15;
    };
    if (adc_value >= 506 & adc_value <= 639) {
        return  5 + (adc_value - 506) * 85 / 133;
    };
    if (adc_value > 639) {
        return 90 + (adc_value - 639) * 10 / 8;
    };

    return 0;
}
```

den3rats, 2025-07-22:
...The numbers themselves in the original table are questionable.
If this is % of the remaining capacity, then an almost horizontal plateau near 3.7V and a strong bend down at  3.3V were expected.
But according to the table, the dependence is almost linear.
As a result, with a remaining capacity of less than 50%, the current calculation will report a lower % than it actually is.

Maybe this is even better, it will motivate you to turn on the charge earlier without waiting for a destructively low level.

diqezit, 2025-07-22:
Thanks for the analysis so I can say that you are absolutely right that the readings will be slightly lower at the lower limit.... However, this is not a mistake, but a conscious design decision made to achieve a balance in my case - relying on the working logic of the initial goshante firmware modified with esp mini hysteresis. Although the dependence looks linear, the table contains 22 calibration points. I think this will be more than enough to adequately track the battery status by voltage.

As for the table, we use 8-bit offsets to store the values in PROGMEM, adding an offset upwards of 491 to avoid storing 16-bit offsets. 
In ENABLE_ADVANCED_BATTERY_LOGIC mode IIR-filter smoothes random oscillations and ADC noise, also added hysteresis.

den3rats, 2025-07-22:
As i understand, battery indicator shows the % of voltage now, not % of capacity.
Yes, it's a simplest way.  

The dependence (ADC -> voltage -> voltage %) is highly linear.  Сode can be simplified even more.
Still need no table for this: 
```
static uint8_t calculateRawPercent(uint16_t adc_value) {
    if (adc_value >= 647) return 100;
    if (adc_value <= 491) return 0;
    return  (adc_value - 491) * 100 / 155;
   
}
```
It would be much cooler to display the remaining capacity in % of mAh. 
But this is a very imprecise heuristic. The dependence is complex, different for different types of batteries, temperature and age of the battery.
Google gives  a lot of bloody different images when you search"lipo voltage vs charge".

den3rats, 2025-07-22:
Trying to convert voltage into charge I got something like this:

```
static const BatteryPointCompressed battery_table[] PROGMEM = {
    // Offset, Percent, (ADC -> Approx. rest of battery capacity (not voltage!!!))
    {156, 100}, // 647 -> 4.15V
    {148,  99}, // 639 -> 4.10V
    {141,  95}, // 632 -> 4.05V
    {132,  88}, // 623 -> 4.00V
    {125,  82}, // 616 -> 3.95V
    {117,  74}, // 608 -> 3.90V
    {109,  67}, // 600 -> 3.85V
    {101,  60}, // 592 -> 3.80V
    { 93,  54}, // 584 -> 3.75V
    { 86,  47}, // 577 -> 3.70V
    { 78,  40}, // 569 -> 3.65V
    { 70,  33}, // 561 -> 3.60V
    { 62,  26}, // 553 -> 3.55V
    { 55,  19}, // 546 -> 3.50V
    { 47,  12}, // 538 -> 3.45V
    { 39,   7}, // 530 -> 3.40V
    { 31,   4}, // 522 -> 3.35V
    { 23,   2}, // 514 -> 3.30V
    { 15,   1}, // 506 -> 3.25V
    {  7,   1}, // 498 -> 3.20V
    {  0,   0}  // 491 -> 3.15V (Safe cut-off)
```

den3rats, 2025-07-23:
Previous table performs this dependence with 21 point:

[image]

But returns to flash savings - we can use only 5 points without critical lost of accuracy:
```
static const BatteryPointCompressed battery_table[] PROGMEM = {
    // Offset, Percent, (ADC -> Approx. rest of battery capacity (not voltage!!!))
    {156, 100}, // 647 -> 4.15V
    {141,  95}, // 632 -> 4.05V
    { 39,   7}, // 530 -> 3.40V
    { 23,   2}, // 514 -> 3.30V
    {  0,   0}  // 491 -> 3.15V (Safe cut-off)
```

[image]