diqezit/ats20_ats_ex #17: Missing shortwave bands

radiola2, 2025-07-22:
The software has no 22, 19 or 11 meter broadcast bands
The existing high shortwave bands are way too wide (no "SW" bands in between).
I guess this was because of code size constraints, but now there is perhaps more space available?

diqezit, 2025-07-23:
Hello. The memory was immediately freed up and was immediately occupied with necessary corrections and additions, so without additional sw sub-bands for now.

radiola2, 2025-07-23:
Sounds great. Waiting for the update to get uploaded.

diqezit, 2025-07-24:
I just wonder if you had the factory default esd assembly on the board or if it was unsoldered?

radiola2, 2025-07-24:
Is there esd protection on the board? I did not unsolder anything. I had a outdoor ~30 meter antenna connected during the thunderstorm.

closed 2025-07-26 by radiola2