diqezit/ats20_ats_ex #20: Settings Menu Timeout Is Too Short

jh4vaj, 2025-07-26:
I tried v5.5. The operation flow has been streamlined, and I find it much easier to use.

In this version, the settings menu automatically closes after a period of inactivity. 
However, the timeout is too short—it often closes while I'm still searching for the desired item.

For other temporary modes, a short timeout is reasonable since the actions are predefined. 
But the settings menu requires more time to browse and locate specific options.

If extending the timeout is difficult due to flash size limitations, 
I would prefer the previous behavior where the menu remained open indefinitely. 
Manually closing the settings menu is rarely an issue.

diqezit, 2025-07-26:
Hello, in order to save memory, I used an existing variable for 3 seconds without creating new ones. I plan to slightly edit the logic in 5.6. Perhaps you have some additional recommendations?

jh4vaj, 2025-07-26:
I think you've already reached an excellent level. For me, it's more than sufficient. 
If I were to point out anything, it would only be minor details.

- The MW band scan is slow due to the 1kHz step.
- The SW subbands don't fully accommodate amateur radio bands.

Neither of these is essential—just things I'd mention if I had to.

closed 2025-07-28 by diqezit