diqezit/ats20_ats_ex #23: v5.6: Compilation error: text section exceeds available space in board

jh4vaj, 2025-07-28:
The compiler said:

Out of a maximum of 30,720 bytes of flash memory, the sketch uses 30,740 bytes (100%).

diqezit, 2025-07-28:
Enable the flags 
-Os -flto -fno-jump-tables

and now I will upload the .hex file. Try flashing it without Arduino IDE.

jh4vaj, 2025-07-28:
It seems that Arduino IDE 2.x doesn’t offer a way to configure those compile options—at least not through the GUI.

So, I used the .hex file you provided. Since it was the “with_bootloader.hex” version, 
I was a bit concerned, but it flashed successfully without any issues. I’m using AVRDUSESS as the programmer.

Regarding the behavior in v5.6, MW band scanning has noticeably improved in speed. 
It locks onto stations reliably, which is excellent.

Scanning with 9 kHz steps works fine, but it doesn’t seem to function properly with 10 kHz steps. 
I suspect that even in 10 kHz step mode, the lower scan frequency might still be fixed at 522 kHz.

Also, I’ve understood how the VCO behaves when adjusted in 13 kHz increments in SSB mode. 
This method seems better.

Thank you very much.

jh4vaj, 2025-07-28:
I’ve confirmed that the font for the “%” symbol has been corrected. Just a brief note to report it.