diqezit/ats20_ats_ex #32: Can't hear the last AM station after re-switching radio

den3rats, 2025-08-03:
Hi!

I noticed this sequence of actions, maybe from version 5*:

- Tuned in to the station.
- Immediately turned the radio off .
- I can't hear the station after turning on.
- If you touch the AVC, the volume returns. 
- But apparently that's not all. Everything is fully normalized only after switching AM->SSB->CW->AM.
I've checked this 7..10 times with different settings, SSB patch loading etc.

My settings as they visible in UI:
```
page1   ATT = 1    SCN = ON    AVC = 90   SMA = 0     SMT =0      DE = 50
page2   BFO = 0    SSM = SNR   SVC = ON   COF = ON    SYN = OFF   SCR = 5
page3   CAP = ON   CPU = 100   BAP = A1   SWU = kHz   RSI = AUT   DIS = OFF
```

https://github.com/user-attachments/assets/9e6df7e3-73fd-43d1-9b85-5cd89a170eec

jh4vaj, 2025-08-03:
This phenomenon was also reproduced with my ATU-20+. 
When I set the ATT to AUT, this phenomenon did not occur.

jh4vaj, 2025-08-03:
Just to be clear, I’m not recommending the use of ATT=AUT. 
I only posted the comment in hopes that it might help identify the cause of the issue.

den3rats, 2025-08-03:
I found version 4.7 in archive.
Yes, the same bug in 4.7

It was a dirty test because I switched to microWire in Tini4kOled to prevent conflicts.
This is in addition to microWire in PU2CLR/SI4735 made earlier.

diqezit, 2025-08-04:
A logical error (regression) was discovered that had been present since version 4.0. The firmware incorrectly assumed that if the attenuator was controlled manually, there was no need to configure the automatic volume control (AVC) parameters at startup. Thank you for your reports and help in improving the firmware.

den3rats, 2025-08-04:
Partial success.
Volume level is OK now.
But the station is also not heard right after switching on.
The same effect from a different angle: Scanning finds a station, for a short moment even a sound flashes (~250ms, transient process in SI473*), but then the sound disappears.

Several alternatives were found to bring AM receiving back to normal:

- switch AM->SSB->CW->AM;
- switch to FM and then to AM back.

diqezit, 2025-08-04:
The problem most likely lies in the initialization of the Soft Mute parameters for AM mode

AM_SOFT_MUTE_MAX_ATTENUATION (maximum attenuation)
AM_SOFT_MUTE_SNR_THRESHOLD (SNR threshold at which Mute is triggered)

Apparently, when the receiver starts up, these parameters are either not sent to the chip at all, or are sent with garbage or incorrect values
Soft Mute system thins that the station has a very low SNR (even if this is not the case) and immediately mutes the sound

When you switch modes, the applyBandConfiguration function is called, which forcibly and correctly sets all these parameters, and the sound appears

den3rats, 2025-08-04:
I tried - there is no effect...

diqezit, 2025-08-04:
```
static void configureAMMode(uint16_t minFreq, uint16_t maxFreq) {
    g_currentMode = AM;
    const Band& current_band = g_bandList[g_bandIndex];
    ModeContext modeCtx = getModeContext();

    g_si4735.setAM(
        minFreq,
        maxFreq,
        current_band.currentFreq,
        g_tabStep[current_band.stepIdxAM]);

    g_si4735.setBandwidth(g_bwAMIdx[current_band.bwIdxAM], 1);

    if (g_modeSettings[MODE_SETTING_AGC][modeCtx] == 0)
        g_si4735.setAvcAmMaxGain(g_modeSettings[MODE_SETTING_AVC][modeCtx]);

    g_si4735.setAmSoftMuteMaxAttenuation(g_modeSettings[MODE_SETTING_SOFT_MUTE][modeCtx]);
    g_si4735.setAMSoftMuteSnrThreshold(g_Settings[SoftMuteThr].param);

    int8_t att_val = g_modeSettings[MODE_SETTING_AGC][modeCtx];
    setAgcHardware(att_val);
}
```

and another one

diqezit, 2025-08-06:
Removed applyband configuration calls from reading reaad eeprom logic; it was unnecessary and should potentially help before initializing the chip after reading valid data from the settings - check this

den3rats, 2025-08-06:
Hello.
Yes, it works OK now.
Thank you!

closed 2025-08-06 by den3rats