diqezit/ats20_ats_ex #37: v6.2 CW L/U receive frequency mismatch

jh4vaj, 2025-08-06:
It seems that the receive frequency differs between CW L and CW U modes. 

For example, when receiving a station at 7020.00 kHz in CW L mode, 
switching to CW U requires tuning to 7019.00 kHz to receive the same station. 

As far as I remember, this issue did not occur in previous versions where CW L/U implementation was complete. 
However, in v6.2, this behavior is present.

diqezit, 2025-08-06:
Thanks - removed the compensation logic so as not to disrupt the CW operation - we will use the original proven operation logic

diqezit, 2025-08-06:
Yeah, that's what the compensation was for.
Let's assume that we care about regression.

 On the display we were showing fake offset data while the chip was getting the correct offset compensation of 500Hz. If you can now hear the difference between lsb/usb, you just need to refine the atomic changes without rebooting the chip via applybandcofiguration.
Did I understand you correctly? Is the CW audio change from usb/lsb working now?

diqezit, 2025-08-06:
Now I understand. Look at 6.2.1. I just updated the commit so as not to release 6.2.2 and confuse users
In general, yes, indeed-the error was caused by a conflict between the manual BFO setting and the automatic CW pitch shift
So we needed to process these two values independently by resetting the manual BFO setting to zero when entering CW mode
This restores the correct auto-compensation behavior from version 6.1.2 and eliminates frequency drift
Thanks again for your help with test!

jh4vaj, 2025-08-06:
Thank you for your prompt response. I’ve tested the new v6.2.1, and it works as expected.

By the way, I felt the CW tone was somewhat low, so I measured it. 
Here’s the procedure I followed:

1. Output 7020.00 kHz from a signal generator.
2. Tune the ATS-20+ to 7020.00 kHz and receive in CW-L mode.
3. Switch to CW-U mode. Since the audible tone frequency differs from that in CW-L mode, 
I fine-tuned the signal generator so that both modes produce the same tone frequency.

With this setup, both CW-L and CW-U produce a tone around 400 Hz (not 500 Hz). 
It may not be a major issue, but I wanted to report it since it caught my attention.

Related to this, I thought it would be nice if users could configure the CW tone frequency. 
It doesn’t have to be arbitrary—just a few selectable options like 500, 600, 700, or 800 Hz would be sufficient.

jh4vaj, 2025-08-07:
I've tested v6.3. The CW pitch control feature is working properly. 
However, the audible tone is consistently 100 Hz lower than the configured value. 
When set to 500 Hz, I hear 400 Hz; at 600 Hz, I hear 500 Hz; at 700 Hz, I hear 600 Hz; and at 800 Hz, I hear 700 Hz. 
Could this be a hardware-specific issue with my ATS-20+?

1. For testing, I came up with a simpler procedure than the one I used yesterday:
2. Output 7020.00 kHz from a signal generator.
3. Tune the ATS-20+ to 7020.00 kHz in CW-L mode (a tone is heard).
4. Switch to CW-U. If the tone pitch differs from CW-L, fine-tune the ATS-20+ receive frequency.
5. Alternate between CW-L and CW-U, adjusting the receive frequency until the tone pitch sounds nearly the same in both modes.
6. Measure the audible tone frequency using a smartphone FFT analyzer app.

This method is easier than changing the signal generator's frequency.

jh4vaj, 2025-08-07:
When CWP is set to 500 and both CW-L and CW-U produce a 400 Hz tone, 
setting BFO to +1 results in a 500 Hz tone in CW-L and a 300 Hz tone in CW-U. 
To make CW-L and CW-U produce the same pitch, the receive frequency must be adjusted. 
And once that adjustment is made, both modes naturally produce a 400 Hz tone again.

If this behavior is to be handled via the BFO setting, 
it would likely require opposite compensation for LSB (CW-L) and USB (CW-U). 
In other words, when BFO is +1, treat it as +100 Hz for LSB and -100 Hz for USB.

I'm not sure if this approach is correct, but based on the observed behavior, it seems to make sense.

diqezit, 2025-08-07:
```
// Sets BFO with user calibration and automatic CW pitch offset
// Si4735 requires an inverted BFO value for sideband selection
static void updateBFO() {
    // Get selected pitch from settings
    uint16_t selected_pitch = pgm_read_word(&cw_pitch_options_hz[g_Settings[CWPitch].param]);

    // Determine automatic CW offset: USB uses a negative offset, LSB a positive one
    int16_t cwOffset = (g_currentMode == CW)
        ? ((g_lastCWMode == USB) ? -selected_pitch : selected_pitch)
        : 0;

    // Determine sideband context for calibration (LSB or USB)
    uint8_t current_sideband = (g_currentMode == CW) ? g_lastCWMode : g_currentMode;

    // Invert calibration sign for USB modes to match observed hardware response
    int8_t sign_multiplier = (current_sideband == USB) ? -1 : 1;
    int16_t bfo_calibration_offset = g_Settings[BFO].param * BFO_CALIBRATION_MULTIPLIER * sign_multiplier;

    // Combine manual tuning, calibration, and pitch offset for the final BFO value
    int16_t finalBfo = g_currentBFO + bfo_calibration_offset + cwOffset;

    g_si4735.setSSBBfo(finalBfo * -1);
}

```

Replace this function on your device, reset the BFO, and change the +1 correction in the settings. I look forward to hearing your results.

jh4vaj, 2025-08-07:
I replaced updateBFO() with the new version and tested it. 
As expected, it works correctly. 
With CWP set to 500 and BFO set to +1, I now hear a 500 Hz tone in both CW-L and CW-U modes. 
Just to be sure, I also tried BFO +2, and both CW-L and CW-U produced a 600 Hz tone as expected. 
Thank you very much.

One question: 
shouldn't the sign handling in BFO calibration also take SSB modes (LSB/USB) into account, not just CW?

diqezit, 2025-08-07:
Yes, it should work in normal SSB mode as well