diqezit/ats20_ats_ex #41: Invisible memories

radiola2, 2025-08-09:
Since the font change every memory item except the last one in the list is invisible.
I made a video that shows it.

https://github.com/user-attachments/assets/a8af808b-6986-48a4-8328-e714a2251b95

diqezit, 2025-08-09:
Hello. There was an error-instead of clearing the line, I cleared the entire screen.
I fixed the clearing: now only the current line is cleared, the list is not erased.
In favorites: mode label on the right, frequencies aligned to the right (1 space before the label).
AM without .00, SSB/CW with .dd;
updated fav_drawLine/fav_drawFreqAMSSB/fav_drawFreqFM/fav_drawModeLabel.

radiola2, 2025-08-09:
The problem is solved. Thanks!